jerabaul29/LoggerWavesInIce_InSituWithIridium #2: stop charging when full battery

jerabaul29, 2019-12-11:
@jvoermans I would think that these lines:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium_private/blob/a2df4c4adb81192624e53e88bad405452b68f7e6/DevelopmentCode/PowerControl/solar_controller.cpp#L83-L86

should ensure that the solar panel does not overcharge the battery. If the voltage is over the max battery level set here:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium_private/blob/a2df4c4adb81192624e53e88bad405452b68f7e6/DevelopmentCode/PowerControl/parameters.h#L49

then something very strange is happening.

I see for possibilities:

- 1 a bug in software somewhere.

- 2 a bug in hardware somewhere.

- 3 an overcharged battery before deployment: is it possible that the battery was wrongly charged / overcharged before being put inside the instrument? In this case, I would expect the voltage to be far too high 'since the beginning' of deployment, and it to fall progressively until the max voltage (3.6V), at which point it may either stay constant (if solar panel is strong enough to keep the battery full), or slowly decrease (if solar panel is not strong enough to keep fully charged).

I would be interested to hear if you can check if scenario 3 is corresponding to your results. If so, then it is not the problem of the logger, but how the battery was charged before deployment. If the battery starts at 3.6V, and only progressively while being deployed gets higher, then this indicates scenario 1 or 2 and I need to debug.

jerabaul29, 2020-08-20:
@jvoermans I have several instruments being deployed in the Arctic now, and the batteries are continuously close to fully charged but without overcharging. So it looks like the on / off switching of the solar panel works fine with these instruments....

There may be one explanation for this: are you using the exact same model of MOSFET in your instruments, or did you choose another model? There are many variations around these MOSFETs, and they have subtle differences. If you confirm that you went for another MOSFET model, if may be the explanation.

jvoermans, 2020-08-20:
I don't think the batteries were overcharged. The charger I supplied with the batteries should have overcharge protection.

I looked up the MOSFETs in my receipts, I got the following:
https://au.rs-online.com/web/p/mosfets/7259322/
https://au.rs-online.com/web/p/mosfets/6715064/

jerabaul29, 2020-08-20:
No worries :) . Yes, it looks like it was probably happening during deployment.

Mmh, ok, this looks like exactly the same components I used. Strange.

jvoermans, 2020-11-03:
@jerabaul29 As a follow-up on the above. I think there might be a bug in the code somewhere regarding battery reading.

Today, I just tested the 'added temperature sensor' sketch on an old logger. Battery is at 3.29V, first message said 4.60V, two hours later 4.59V and half hour after 2.98V.

Also, one of the two currently deployed in Antarctica has similar issue regarding battery voltage. It was around 2.5V whole time and still working. They brought it back to the station but battery was fully charged according to the station's electrician (3.5V). Deployed again and stays around 2.5V still working. Any ideas what to test?

jerabaul29, 2020-11-03:
Ok, many thanks for the update! This is strange, as I have never observed the problem myself with my instrument. Wonder if we use a different component somewhere without being aware about it. Or if we are operating in different regimes (maybe this bug appears only when the batteries are completely full and saturate, and this is not something I have observed myself). But then we know that we should just "safely ignore this information".

So it looks like an intermittent hardware / software issue. These can be really quite hard to debug, specially when I myself cannot reproduce the issue... I think it is best to just ignore this problem. Anyways we know the instruments work correctly in reality, which is great, and the most important. This will get better the day I do a full re-design of the instruments (trying to get a project about this, I will keep you posted and invite you to join the application if I am allowed to proceed by my boss :) ).

closed 2021-11-06 by jerabaul29