jerabaul29/LoggerWavesInIce_InSituWithIridium #3: Power Control

jvoermans, 2019-12-18:
Hi Jean,

Could you tell me how to upload the PowerControl code to the ATMega328P?
Do I just fix it to a Arduino Uno and upload?

Joey

jerabaul29, 2019-12-18:
Hi Joey,

Yes, exactly. The only things you need to be careful about is 1) do not discharge static electricity on the chip (ie touch metal before), 2) be careful to not bend the legs of the chip, 3) insert the chip in the right direction (i.e. check the position of the small circular mark).

Let me know if you have any problems.

Btw, is your instrument still working in the Antarctic? :)

jvoermans, 2019-12-18:
Ok, I did the following:

I directly put the ATMega328 on the Arduino Uno, uploaded the sketch. I put it on one of the spare loggers, but when connecting to power, it just keeps turning on every time I connect it, so I'm doing something wrong.

Should I do something like they do here?:
https://www.instructables.com/id/Build-your-own-Arduino-Bare-Bone-System/

FYI: the ship was still around the loggers, so they simply disconnected and reconnected the battery and it is transmitting again. The ice on which they are deployed is however protected from the Southern ocean by a second strip of ice, so there are basically now waves, but if the batteries are kept fully charged, we just have to wait and hope the ice floes don't melt ;)

jerabaul29, 2019-12-18:
Ok. Did you follow the instructions here?

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/master/notes_building_Oct_2019.md

"
code for the Arduino barebone
power controller: using commit b76f6a5 (current master IS BROKEN), upload the code in the PowerControl folder on the barebone Arduino (for example, using an Arduino Uno to program the chip). Check the parameters and that you are ready for deployment, not test.

go back to master in the code
"

Yes, this is the right technique. You can just use the Arduino IDE to compile, or platformio.

Ok. Do you mean there are 1 or 2 loggers active now? Is the logger that had a problem working again? Or still only 1 working?

jerabaul29, 2019-12-18:
So:

got checkout b76f6a5

Then upload

Then 

git checkout master

You can check the status of the git tracking by 'gut status'. If you are confused I can create a separate folder with the code, may be easier to find, do it adap I have internet access to my computer.

jerabaul29, 2019-12-19:
@jvoermans if you are a bit less familiar with git commands you can just use the power controller code here:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/tree/master/DevelopmentCode/PowerControl_commitb76f6a5

Can you confirm to me that it works?

jvoermans, 2019-12-19:
Yes that worked! Thanks. That is one software problem down!

And yes, now both loggers are active again in the Antarctic! Not sure what the problem was, if reconnecting the battery solved it, that means that it is probably a hardware thing.

jerabaul29, 2019-12-19:
Mmmh, very strange, should not be able to happen. Maybe some bad contact of some sort between the battery and the logger that was solved by plugging in and out, or some bad soldering close to the plug that adjusted a bit with playing with the connector. Hope it holds. Good both work again :)

closed 2019-12-19 by jerabaul29

jerabaul29, 2019-12-19:
PS @jvoermans : make sure you adapt the wakeup parameters for wakeup frequency to your needs :)

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/6cec5acdfae4ed4b094409de874d25b971efb4b8/DevelopmentCode/PowerControl_commitb76f6a5/parameters.h#L52-L68