jerabaul29/LoggerWavesInIce_InSituWithIridium #4: Arduino PlatformIO

jvoermans, 2019-12-18:
Hi Jean,

I installed VScode with PlatformIO extension on Linux system. I tried to upload the code onto the Arduino, but get some errors. Any suggestions? Below is terminal messages

> Executing task in folder ArduinoMega: platformio run <

Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR 2.0.0 > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduino-avr 5.0.0
Converting src.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <EEPROM> 2.0
|-- <Adafruit GPS Library> 1.3.0
|   |-- <SPI> 1.0
|   |-- <SoftwareSerial> 1.0
|   |-- <Wire> 1.0
|-- <SoftwareSerial> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SD>
|   |-- <SPI> 1.0
Building in release mode
Compiling .pio/build/megaatmega2560/src/src.ino.cpp.o
In file included from src/IridiumManager.h:2:0,
                 from /home/joey/Documents/UniMelb/Ice_Logger/ArduinoMega/src/src.ino:44:
src/IridiumSBD.h: In constructor 'IridiumSBD::IridiumSBD(Stream&, int)':
src/IridiumSBD.h:161:18: warning: 'IridiumSBD::lastPowerOnTime' will be initialized after [-Wreorder]
    unsigned long lastPowerOnTime;
                  ^
src/IridiumSBD.h:145:15: warning:   'IridiumSBD::StreamShim IridiumSBD::diag' [-Wreorder]
    StreamShim diag;
               ^
src/IridiumSBD.h:88:4: warning:   when initialized here [-Wreorder]
    IridiumSBD(Stream &str, int sleepPinNo = -1) :
    ^
src/IridiumSBD.h: In constructor 'IridiumSBD::StreamShim::StreamShim(IridiumSBD*, bool)':
src/IridiumSBD.h:123:19: warning: 'IridiumSBD::StreamShim::isbd' will be initialized after [-Wreorder]
       IridiumSBD *isbd;
                   ^
src/IridiumSBD.h:122:12: warning:   'bool IridiumSBD::StreamShim::diags' [-Wreorder]
       bool diags;
            ^
src/IridiumSBD.h:126:7: warning:   when initialized here [-Wreorder]
       StreamShim(IridiumSBD *isbd, bool d) : isbd(isbd), diags(d) {}
       ^
In file included from src/EEPROM_interaction.h:2:0,
                 from src/SDManager.h:3,
                 from src/GPSManager.h:6,
                 from /home/joey/Documents/UniMelb/Ice_Logger/ArduinoMega/src/src.ino:40:
/home/joey/Documents/UniMelb/Ice_Logger/ArduinoMega/src/src.ino: At top level:
/home/joey/.platformio/packages/framework-arduino-avr/libraries/EEPROM/src/EEPROM.h:145:20: warning: 'EEPROM' defined but not used [-Wunused-variable]
 static EEPROMClass EEPROM;
                    ^
Linking .pio/build/megaatmega2560/firmware.elf
lto1: internal compiler error: in lto_output_varpool_node, at lto-cgraph.c:624
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://www.atmel.com> for instructions.
lto-wrapper: fatal error: avr-g++ returned 1 exit status
compilation terminated.
/home/joey/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
*** [.pio/build/megaatmega2560/firmware.elf] Error 1
================================================================================ [FAILED] Took 0.95 seconds ================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

jerabaul29, 2019-12-18:
Mmh, strange, I have never seen something like that myself. As you have seen the code compiled nicely on my laptop. Have you tried googling the issue / asking for help online / filling in the bug report?

jerabaul29, 2019-12-18:
(I am traveling at a conference in China, not easy for me to help debug just now. What is the deadline for assembly of your instrument?)

jerabaul29, 2019-12-18:
Which code are you trying to compile? This one right?

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/tree/master/DevelopmentCode/ArduinoMega/src

You should compile from the src.ino if I remember well.

Worst case if you cannot compile, can you try to use the binary in the .build folder and upload it on the mega without recompiling? You should look for instructions if / how this can be done on platformio (if needed ask for help / open an issue).

jvoermans, 2019-12-18:
Deadline is the end of the month.

I find it a bit confusing. I tried a simple led blink sketch, which uploads to the arduino uno and mega fine through the arduinoIDE. Through platformioIO, it only works for the uno, but not the mega. Then you get the error message:
avrdude: stk500_recv(): programmer is not responding

Seems to be very generic error message, haven't solved that one yet.

I prefer to use platformio as it seems easier to compile, but can I do the same with the arduinoIDE if I restructure the folders and use the extended buffer again?

jerabaul29, 2019-12-18:
Mmh, ok. It is often quite some 'emotions' with assembling hardware on your own for the first time with tight deadlines, but I can do my best to help.

It sounds like there is a very fundamental step in the setup of platformio for the mega that must have a problem.

- are you sure you have installed all necessary packages / dependencies for platformio working with the Arduino Mega?

- are you sure you have selected the right parameters in the code setup so that paltformio us aware it should talk to a mega? There is at least 1 maybe more places in platformio where you have to select Mega from a menu. Also, there may be different kinds of mega, I think if I remember well you have to select mega 2560 or something like that.

jerabaul29, 2019-12-19:
@jvoermans just to illustrate how things look like on my computer (see the options on the lower bar):

[image]

One more thing: the first time I tried to compile I got an error message for some strange reason, a bit in the same kind of you. But trying to compile one more time, without changing anything, fixed the problem somehow. Can you try to launch compilation a couple of times? You can try first to compile, not upload.

jvoermans, 2019-12-19:
I can open the src.ini file, but do get the following popup message:

C/C++ IntelliSense service does not support .INO files. It might lead to the spurious problems with code completion, linting, and debugging. Please convert .INO sketch into the valid .CPP file.

I already tried compiling multiple times, but get same error message. I think it is not recognizing the board somehow.
[image]

jerabaul29, 2019-12-19:
- The intellisense error message is not a problem. This just means that the intelligent code completion will not work with .ino files, and that it is recommended to change it into a .cpp file instead.

- Have you tried to troubleshoot with help from the internet? Can this help?

https://github.com/platformio/platform-atmelavr/issues/65

Are you using linux or windows? This looks like windows from your command line output. Many things that are open source do not like windows and break on it. Can you try with a linux computer instead? 

- Have you tried to report the bug as suggested? This is related seems to a compiler that was broken. People have god some help on that before:

https://github.com/arduino/arduino-builder/issues/332

https://github.com/arduino/Arduino/issues/7949

jerabaul29, 2019-12-19:
@jvoermans another solution is that you use the Arduino IDE if you cannot get Platformio to work. This should be fine too. The only thing is that you need to make sure that in this case you extend the size of the Arduino Mega buffers sizes, similar to what I do in platformio here:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/6cec5acdfae4ed4b094409de874d25b971efb4b8/DevelopmentCode/ArduinoMega/platformio.ini#L17

This should be possible to perform following these instructions:

https://forum.arduino.cc/index.php?topic=399751.0

or here:

https://forum.arduino.cc/index.php?topic=538583.0

You will know very quickly if you get it to work or not. If not, the logging of the VN100 will not work. Also, you can check the size of the compiled sketch, and the memory use estimate to check the buffer size was changed successfully:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/master/notes_building_Oct_2019.md

"

    upload the code (from ArduinoMega, current master, for example using commit 39c30775849a). Make sure that you well use the extended buffer compilation flags (see the platformio.ini parameters folder). The sizes of the data and program should read something in this kind:

DATA:    [========  ]  81.5% (used 6673 bytes from 8192 bytes)
PROGRAM: [=         ]  10.9% (used 27732 bytes from 253952 bytes)

"

Make sure that you follow the instructions of the build notes from october and that you initialize the EEPROM first in particular :)

jvoermans, 2019-12-20:
Nope. Also Linux same issue. A simple sketch through platformio can be uploaded on the mega2560. 

I tried now to do all with the ArduinoIDE. I managed all the libraries, but it says it cannot find the 'parameters.h' file, while it is clearly in the folder. 

Arduino: 1.8.4 (Linux), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch/HighSpeedUSBOutputController.cpp:2:0:
/home/joey/Arduino/libraries/DebugMacros_Lib/DebugMacros.h:2:24: fatal error: parameters.h: No such file or directory
 #include "parameters.h"
                        ^
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

Does this mean I have to add the 'parameters.h' header in the DebugMacros library folder? that sounds a bit weird to me, if I do, I get quite a few other errors

jerabaul29, 2019-12-21:
@jvoermans it should work now:

- I have put a separate folder here that should compile under arduino ide:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/tree/master/DevelopmentCode/ArduinoMega_commit6cec5acdf/src

# solution 1: compile locally

- You can compile / upload yourself if you want through the Arduino IDE. For this, you need to update the size of the buffers first. For this, you should find the location of your arduino core, and then go to:

**locations_of_install/arduino-1.8.10/hardware/arduino/avr/cores/arduino**

and there modify the *HardwareSerial.h* to:

```
#if !defined(SERIAL_TX_BUFFER_SIZE)
#if ((RAMEND - RAMSTART) < 1023)
#define SERIAL_TX_BUFFER_SIZE 16
#else
#define SERIAL_TX_BUFFER_SIZE 512
#endif
#endif
#if !defined(SERIAL_RX_BUFFER_SIZE)
#if ((RAMEND - RAMSTART) < 1023)
#define SERIAL_RX_BUFFER_SIZE 16
#else
#define SERIAL_RX_BUFFER_SIZE 512
#endif
#endif
```

(i.e. 2 values of 64 changed to 512). In this case, the compilation should give something like:

```
Sketch uses 28672 bytes (11%) of program storage space. Maximum is 253952 bytes.
Global variables use 6949 bytes (84%) of dynamic memory, leaving 1243 bytes for local variables. Maximum is 8192 bytes.
Low memory available, stability problems may occur.
```

This confirms that a lot of RAM is used because the buffers have been extended.

# solution 2: upload the hex I compiled and put on the repo

- Another solution is to use directly the .hex file I have uploaded on the repo. Not sure though if you should use:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/master/DevelopmentCode/ArduinoMega_commit6cec5acdf/src/src.ino.mega.hex

or:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/master/DevelopmentCode/ArduinoMega_commit6cec5acdf/src/src.ino.with_bootloader.mega.hex

and to upload it on your arduino mega using XLoader (I have not tried myself before, but should be ok).

How I generated the .hex files:

https://arduino.stackexchange.com/questions/48431/how-to-get-the-firmware-hex-file-from-a-ino-file-containing-the-code

How it should be possible to upload with XLoader:

http://www.hobbytronics.co.uk/arduino-xloader

jvoermans, 2019-12-21:
Thanks Jean for your help. I used the first solution. First upload the initializeEEPROM, then use the extended buffer core to upload the src. Get almost the exact numbers you get:

Sketch uses 28678 bytes (11%) of program storage space. Maximum is 253952 bytes.
Global variables use 6949 bytes (84%) of dynamic memory, leaving 1243 bytes for local variables. Maximum is 8192 bytes.
Low memory available, stability problems may occur.

I'll test tomorrow at uni.

Could you tell me what is different with this folder compared to the other? The compiling goes almost instantly, is it actually compiling or does it bypass and uses info from the hex file?

jerabaul29, 2019-12-21:
Perfect, great that it works!

I think it does do the compilation, no tricks here :) I just fixed a few things in the source code. Keep me informed if it works.

jerabaul29, 2019-12-24:
This seems to work, closing.

closed 2019-12-24 by jerabaul29