jerabaul29/LoggerWavesInIce_InSituWithIridium #5: RPi image

jvoermans, 2019-12-18:
Can I just put the complete image directly (https://folk.uio.no/jeanra/Research/img_0208_Rpi_2Bv11.img) onto the SD card with the Startup Disk Creator, or do I have to complete all the setup steps at 'master/DevelopmentCode/RPi_Side' as well?

jerabaul29, 2019-12-18:
It depends on the size of you SD card. If the SD card and the image have the same size (or nearly), see option 1 here:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/master/DevelopmentCode/Preparing_RPi/how_to_prepare_RPi.txt

Otherwise see here, RPi part:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/master/notes_building_Oct_2019.md

In general you should read in details this last file to make sure you do all steps correctly

jvoermans, 2019-12-18:
On another note, I think I bought the wrong RPi, it is the B+ v1.2, seems to have 512MB RAM.

And as per 'possibility 1' of the first (using 16GB sd card), is that the sameasjustusing startup disk creator? (sorry, I'm very new to all this programming)

jerabaul29, 2019-12-18:
Mmmh, RPi v1.2 may be too weak. Can you order some of the same as you had with you to UiO (just check which model you had on your last order / receipt if possible).

Yes. If your SD is 16GB, upload with DD but no resizing. If larger (may be a good idea for long deployment), use the resizing method.

closed 2020-01-05 by jvoermans