jerabaul29/LoggerWavesInIce_InSituWithIridium #6: building instruments

jerabaul29, 2019-12-20:
@jvoermans how are things going?

- power controller is ok, right?
- how is it going with RPi image?
- how is it going with Mega code upload?

:)

jvoermans, 2019-12-24:
I tried to start the loggers we finished in Oslo, but they act differently. I think it is the battery I used (I connected it to 2x AA batteries). Measuring the voltage across the step-up converter seems alright, but when the logger is supposed to log, the GPS keeps blinking for 1-2 minutes, then the SD card finally started recording, and at the end of the 25 minutes of recording, it started blinking with lower frequency. After 25 min it stopped (as usual) but the RPi never started. I checked the file, the accelerometer data seems fine, and all in the file, but the last part a lot of zeros popup and infrequently it gives accelerometer data. The fact that it starts recording (irrespective of the RPi doing its work), does that mean that the RPi is successfully identified by the arduino, that is, would it start recording when the RPi is not attached?

Do you think the battery is the issue?

When I use the 'logger in ice' (without iridium), and connect to the computer, everything works normal. So I think its the battery. It's Christmas here, so can't get a proper battery right now to test, so have to wait for that unfortunately. If you ever experienced something like the described, let me know.

jvoermans, 2019-12-24:
Also, I used a 1000uF (16V) rather than 1000uF (10V) capacitor and a 1uF (63V) instead of 1uF (50V), (electro guys said it wouldn't matter for the Arduino). Would you agree?

jerabaul29, 2019-12-24:
- The capacitors make no difference at all, agree with your engineers :)

- Yes, I think your batteries are too weak. Can you try 3 X AA in series? The problem is that AA nominal voltage is 1.5V, actually closer to 1.4V, so 2.8V total. In addition when you add voltage drop due to internal resistance you may be down to maybe 2.75 or 2.7 V. In this case, the logger may automatically shut off when a large load (RPi) wakes up, and / or the voltage may drop too low and things start to fail. I think this may also explain for your strange data. Also, those batteries have very little capacity, so they may be quite depleted (and then ever lower voltage) by the time the RPi wakes up. When the RPi is on the intensity draw on the batteries may be up to 1.5 or 2 A, which is a lot for so small cells.

- which RPi model do you use? The same you had to Oslo? Do you see some of the LEDs of the RPi turning on at some point at all? 

- Can you use a lab power supply at 3.1V as input to you logger prototype rather than batteries?

jerabaul29, 2019-12-24:
PS: still getting data from Antarctica?

Merry Christmas! :)

jvoermans, 2019-12-24:
Yes, they are still working and still full battery. Still no waves though :p but at this rate they can survive for a long time.

I'll ask if they have bench power supply, how many amps should I use with 3.1V?

Jij ook vrolijk kerstfeest!

jerabaul29, 2019-12-24:
Excellent, so happy with the successful deployment, that really makes me happy 🎉🎉 Hope for sun to keep the batteries happy, no snow falls to not cover the antenna, and some waves some time soon!

You should use a lab supply, put it in 'set voltage' at 3.1V, and put a max intensity at 5A or something like this. I.e. it should be voltage regulated without any intensity limit in practice (high enough to not limit).

jvoermans, 2019-12-30:
Surprisingly, still not working!

I used a bench power supply, 3.3V, max 5A.

This is what happens:
-everything goes normal, after few reconnections of battery, logging occurs rather quickly as it should be, current drawn about 0.2A for 25 min.
-Then for 5-6 minutes, nothing really happens with the RPi, only GPS breakout blinks normally, and the current stays around 0.3A, and every 20 seconds, it increases in 1-3 seconds to 0.5A after which it drops back to 0.3A
-After that, the green light starts burning of the RPi, this takes 2 minutes.
-Then RPi doesn't do anything, only GPS breakout blinks for about 10 seconds and the sequence is over.

One thing, I accidentally mixed up the SD-cards of the RPi's. Is that an issue? I rotated all three SDcards of the RPi to see if anything changes but they all lead to the same result.

Also, accelerometer data files are fine on the SDcard of the breakout. They are not present on the RPi, so they were not transferred to the RPi.

Also, is it normal that the logger can log even without the RPi being attached? This is not the case for the Iridium, if I detach that one, it won't start logging...

jerabaul29, 2019-12-30:
Ok.

This sounds like everything is fine until the RPi and the Mega should talk. Somehow they do not manage to talk to each other, this is when the sequence is aborted.

Are you well connecting the RPi and the Mega with an USB cable? Have you checked that the USB cable is 'good' (can you upload an Arduino sketch through it for example? Some USB cables are 'dummies' that only transfer power, no data).

Yes, it is expected that the loggers work without RPi, bug not without Iridium.

Are you sure you correctly burned the image on the SD card following the instructions? Can you just give power to a RPi through the mini USB and let it boot, and check that both the green and red lights blink then?

Which model of RPi are you using? Is it the same as you used when in Oslo?

Eiserklaus, 2019-12-30:
Hi,

I have no hardware here to test anything (as mentioned I left Labmaker) but my 2 cents: intermittent fails of Raspis are most often power issues. I experienced fails in one of my projects each time the USB wifi tried to send due to insufficient power. So my idea would be to power the Arduino separately (if it is not) and try perhaps a large capacitor in the power supply to bridge peak power consumption. Same considerations may hold true for other power intensive components like the Iridium.
Also removing all useless (power-) LEDs on the Raspi and Arduino may help to some extent.

jerabaul29, 2019-12-30:
Yes, power is always a possible issue. But we know that with steady 3V input this setup should work fine... Wonder if there is another problem: RPi version, SD card preparation, faulty USB cable or other.

jerabaul29, 2019-12-31:
Ok, looks like this does is a power issue encountered when using a lab power supply instead of a battery, strange. Probably the step up converter does not like having an input with any ripples or something like that. Will investigate with pololu. @jvoermans let us know when you have sent the message to them.

jerabaul29, 2020-01-02:
@jvoermans can you confirm this was due to wrong settings of your power supply, and that the new series of instruments you built and programmed works fine? If this is well the case feel free to close.

jvoermans, 2020-01-02:
Haven't been able to test the settings of the power supply yet. Most likely, problem is the power supply, increasing the voltage of the bench power supply to 4.5V led to a steady input voltage of 5.2V on the PCB (both with Arduino active and inactive).

jerabaul29, 2020-01-02:
There should be a way to set the max current of the supply and increase it to a few amps, enough to power at 3.2V. Can you ask your engineers for help? Should do the trick.

jvoermans, 2020-01-02:
I think the max current is standard at 5.2A. When you turn on the supply, it shows briefly the current settings. But I have to double check. I'll get back on this.

jerabaul29, 2020-01-02:
5.2 A should be enough at 3.2V. I would think there is a restriction on it now. May be wrong though.