jerabaul29/LoggerWavesInIce_InSituWithIridium #7: Iridium transmission

jvoermans, 2020-01-05:
@jerabaul29 
is the RPi or the Arduino responsible for the final transmission of data? i.e. if I want to send temperature through Iridium as well, perhaps easiest to do with the GPS data transmission, what files should I start?

jerabaul29, 2020-01-05:
The Arduino is responsible. Yes, it is this file. I let you look at the code to get started. Basically, you need to 1) write the day in the TX buffer of the Iridium 2) ask to send.

Eiserklaus, 2020-01-08:
Hi,

Hi, would this module be of any value for your research:
https://www.taoglas.com/product/edge-locate/

jerabaul29, 2020-01-08:
Hi,

This could be of interest for measuring very low frequency waves, but I think the IMU is a bit better in our case (we manage to measure waves of a few mms, but this has a cm resolution only).

closed 2021-11-06 by jerabaul29