jerabaul29/LoggerWavesInIce_InSituWithIridium #16: ArduinoMega

jvoermans, 2020-10-08:
@jerabaul29 I've been trying to get the code on the ArduinoMega, this time I attempted through VSC again, but there is an error that pops up everytime:
C:\Users\jvoermans\.platformio\lib\Adafruit GPS Library_ID20\src/Adafruit_GPS.h:58:28: fatal error: SoftwareSerial.h: No such file or directory

jerabaul29, 2020-10-08:
You need to put the Software Serial in the lib folder. Just put the whole content of

https://github.com/arduino/ArduinoCore-avr/tree/master/libraries/SoftwareSerial

In lib in the project and let me know if it works.

jvoermans, 2020-10-08:
I added SoftwareSerial folder and the h/cpp files in the lib folder, but still get the same error. See figure attached.

[image]

jerabaul29, 2020-10-08:
Mmh, strange, are you sure that the whole Software serial is in the lib folder? I.e. do you well have the liv/SoftwareSerial/src and everything there?

# possible fix 1

A possibility is that you may need to change to ```#include "SoftwareSerial.h"``` here:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/7bb2ff9ef0ba867dc73101b6fbd4b49d061ea94f/DevelopmentCode/ArduinoMega/src/GPSManager.h#L5

# possible fix 2

Otherwise, can you follow the advice, and install directly through the platformio software dependency manager as suggested in the output?

# best to "keep it as it was working before"

But you were able to compile this before as you already built some loggers, right? Can you just use the exact same method as what you used and was working before? I set this up a long time ago, I may have been a bit sloppy with the dependencies management, I will try to be more consistent on newer code.

jvoermans, 2020-10-08:
Yes, uploading through the Arduino IDE  you provided works absolutely fine. But I wanted to move to VSC so tried again.

The above didn't work either, however, I added 'lib_ldf_mode = deep' to the platformio.ini file did the trick
https://community.platformio.org/t/build-fails-because-missing-library-include-arduino-framework/14146/2

I end up with a different error now, it looks quite familiar, I think this is where I ended last time with the errors 'ld.exe: error: lto-wrapper failed'. I'll just working from the IDE then for this one

jerabaul29, 2020-10-09:
Ok, sounds good! Yes it works on my Plarformio, not sure why not on yours. I suppose I installed some libraries globally on my machine. I will not do this in the future ,but always put stuff in the lib folder, this way it will be fully reproducible :) .

Ok, good luck!