jerabaul29/LoggerWavesInIce_InSituWithIridium #17: Extended Buffer

jvoermans, 2020-10-08:
To get back to this topic. I was re-attempting to get the extended buffer working on my work laptop again (Windows 10), and finally figured out why it wasn't working. Apparently, Windows 10 has a hidden Arduino folder in 'AppData'. If you go to File-->Preferences in the Arduino IDE, there is at the bottom a location where the Arduino 'preferences' are stored. Here is another folder with all the boards, cores etc. I'm sure you can change the settings of that somewhere, but changing it in the 'AppData' folder works fine for me. Anyway, not relevant for you on Linux, but perhaps good to document.

I had a question regarding which files you have to change the buffers. Is it just in the HardwareSerial.h, as you stated earlier:
https://github.com/jerabaul29/LoggerWavesInIce/tree/master/ArduinoMegaExtendedSerialBuffers

Or also in the 'USBAPI.h' as stated in another source:
https://www.hobbytronics.co.uk/arduino-serial-buffer-size

jerabaul29, 2020-10-08:
I do not have access to a Windows computer and cannot help debug on Windows platforms.

Do you have access to a Linux computer? That would be the simplest. If not, can you ask Alexander to buy for your lab a Thinkpad T490 or similar, put Ubuntu 20.04 on it, and use it as "the lab's Ubuntu machine"? It is a pain to write code on Linux and make sure it works on Windows...

My post should be a good description of what I did. That was a long time ago though, I do not remember everything.

For this specific project: you had a successful method for uploading the code on the Mega as you built several instruments last year, right? :) Can you continue with this method (the "don t fix it if it s not broken", apologies ^^). This code starts to be quite old, I do not really want to spend to much time supporting it, would prefer to re-develop something much cleaner.

jerabaul29, 2020-10-08:
See this:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/issues/4

jvoermans, 2020-10-08:
All good

closed 2020-10-08 by jvoermans