jerabaul29/LoggerWavesInIce_InSituWithIridium #18: Add sensors

jvoermans, 2020-10-30:
@jerabaul29 Hi Jean, trying to explain my approach on how to change the sketch to have the first Iridium message include mean temperature from a  sensor. Could you guide me in whether this is ok to do?

(1) measure temperature directly after logging has finished. So some lines to record temperature over, say 2 minutes, comes in between these lines:

https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/7bb2ff9ef0ba867dc73101b6fbd4b49d061ea94f/DevelopmentCode/ArduinoMega_commit6cec5acdf/src/src.ino#L120-L123

(2) Store the mean temperature in a variable, for example: TEMP_AVE

(3) Add lines to the 'message to be send'. Probably copying and editing this:
https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/7bb2ff9ef0ba867dc73101b6fbd4b49d061ea94f/DevelopmentCode/ArduinoMega_commit6cec5acdf/src/IridiumManager.cpp#L112-L125
If the temperature is stored under a variable "TEMP_AVE", can I just say:
String TEMP_AVE_string = String{TEMP_AVE};  ?

(4) Add something similar as this in the header file:
https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/7bb2ff9ef0ba867dc73101b6fbd4b49d061ea94f/DevelopmentCode/ArduinoMega_commit6cec5acdf/src/IridiumManager.h#L64-L68

Am I forgetting something? while this increases the bytes of the message, there is no problem technically I assume?

Ideally, a datafile is created on the sd card to store sensor data as well, i.e., at each recording, append one file with the average temperature + corresponding file name. I'm sure this is possible?

jerabaul29, 2020-10-30:
Yes, exactly, that should be all what you need to do. A few notes:

- you should develop your code on a new branch, forked from the commit ```39c30775849a```. I created just a branch just now, so you can work directly on it :) https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/tree/feat/add-temp-meas

- remember to regularly reset the watchdog if you have code running some logging for a long amount of time. The command is simple: ```wdt_reset();```

- you will also need to tune a bit the python code for deciphering the iridium messages, but that should be fine.

Yes, there should be plenty of space left on the status message for adding this information.

One more thing: I do not really think that you need to take the average over such a long periods. If the measuring time is 10 ms, maybe it should be enough to average over 100 measurements ie 1 second?

jvoermans, 2020-10-30:
Thanks. You might have explained already, but what is the watchdog exactly?

jerabaul29, 2020-10-30:
It resets and reboots the board if not called regularly. The ```wdt_reset()``` command is what resets it, confirming that the program is running well without hanging.

https://circuits4you.com/2018/01/24/tutorial-on-arduino-watchdog-timer-setup/

jvoermans, 2020-11-03:
Ok I was able to get temperature in the GPS Iridium message. In a separate clone of the repo I did last week.

To work from the branch instead, I cloned the 'feat/add-temp-meas' branch. I added the word 'test' in the readme file of the branch, and tried to push it, but I get a 'unable to access' error. Do I need permission from you I guess to push it?

jerabaul29, 2020-11-03:
That sounds excellent, congratulations!

I think you may need actually to perform a pull request since this is a public repo. See:

https://docs.github.com/en/free-pro-team@latest/github/collaborating-with-issues-and-pull-requests/about-pull-requests

https://docs.github.com/en/free-pro-team@latest/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request

https://yangsu.github.io/pull-request-tutorial/

Basically when there is a public repo, there are not always collaborators as in a private repo, and to make it possible for everybody to participate but for the owner to still have control, there is this pull request system. It works very well :) 

Let me know if anything is unclear.

jerabaul29, 2020-11-03:
(note: you may need to create a public fork before you can do a pull request; may be easier to just fork first, then copy paste your changes anew, and then do a pull request).

closed 2021-11-06 by jerabaul29