jerabaul29/LoggerWavesInIce_InSituWithIridium #20: 3.3V pin

jvoermans, 2020-12-08:
@jerabaul29 I think the 3.3V pin is not connected to anything right? So I can use it for 3.3V sensors?

jerabaul29, 2020-12-08:
Mmh, which pin? Arduino Due 3.3V? Yes, it should be a stable and safe 3.3V source. Why? :)

jvoermans, 2020-12-08:
No, the mega 3.3V pin. We solder it but I don't think I see it going anywhere. The temperature sensors of blue robotics each need an I2C level converter for 5V to 3.3V (supply voltage 5V is fine, but logic needs maximum 3.6V), so this problem is for mega only. But the converters of bluerobotics are relatively big and expensive compared to others as they have two easy click sockets which I won't use:
https://bluerobotics.com/store/comm-control-power/tether-interface/level-converter-r1/

Sparkfun has this one:
https://www.sparkfun.com/products/12009

but it needs a reference Voltage of 3.3V.

jerabaul29, 2020-12-09:
Yes, sorry, I was answering too many issues in bulk and had not noticed this was on another repo :)

So basically:

- yes, feel free to use the 3.3V as a powering pin. It is obtained by a low dropout regulator tapping on the 5V rail. It is perfectly fine to use it :) 

- I think the sparkfun adapter looks very good. Should work out of the box without problem. Make sure to test it in details :) 

Let me know if you have any problems.

closed 2021-11-06 by jerabaul29