jerabaul29/LoggerWavesInIce_InSituWithIridium #23: using 32GB card image

jerabaul29, 2021-01-19:
@jvoermans let's discuss here how it goes with using the 32GB card image.

jvoermans, 2021-01-20:
initially solved. Will do test today...

jvoermans, 2021-01-20:
You remember when the instrument was deployed in the Antarctic and only GPS messages came through, not the spectra? I first thought the SD card of the RPi was full (which is why I wanted to make sure this time I increased the size). I now actually think the battery is starting to get empty and cannot handle the RPi anymore, draws too much current, and voltage drops to about 3.6V instead of 5.1-5.2V. So, until the RPi is started, everything works fine, but from there on not anymore...

jvoermans, 2021-01-21:
Works!

jvoermans, 2021-01-21:
Am I missing the 'close' button somewhere? Or are you the only one able to do so? ;)

jerabaul29, 2021-01-21:
Mmh no you should be able to close too. See the button left to Comment :) .

Great that it works :) .

jvoermans, 2021-01-22:
That's what I though. I might be crazy...
[image]

jerabaul29, 2021-01-22:
Really weird. I will close then.

closed 2021-01-22 by jerabaul29