jerabaul29/LoggerWavesInIce_InSituWithIridium #24: Low memory

jvoermans, 2021-01-22:
@jerabaul29 Is it possible that the low memory (i.e. ~90% of dynamic memory is used) is a problem when adding more sensors? I've added parts of scripts that work fine on an barebone Mega, but cause issues on the Logger. The sketch compiles fine, the logger starts, but does not record.

jerabaul29, 2021-01-23:
Mmmh, it is possible. I never thought about adding so many sensors... Is it ok to limit just a bit the number of sensors so that things work?

On the instruments V3 that I think of I will use a board with much, much more memory that has become available only very recently. This will solve this kind of problems...

I think you may consider which sensors to put on this Vs the new Due loggers. For example maybe the sonar would be better on the due, recording both distance and full ping profile as I am working on? ... I have some compilation problems at the moment but will look at it this WE.

jvoermans, 2021-01-24:
Ah ok. Yes, I guess pressure sensor can go. It is least important to the experiments. Although Meteorologists are especially interested in that...

I find that 7493 bytes of dynamic memory is more or less the threshold, though it seems to differ a bit. I think I'm short about 10-20 bytes to get all sensors in. Btw, can I change the String below into: String(F("B,\n") ?
That would give me 4 bytes... Or does this mess things up?

For example,:
https://github.com/jerabaul29/LoggerWavesInIce_InSituWithIridium/blob/7bb2ff9ef0ba867dc73101b6fbd4b49d061ea94f/DevelopmentCode/ArduinoMega_commit6cec5acdf/src/VN100Manager.cpp#L122

jerabaul29, 2021-01-24:
No I do not think you should change that, as this is used by the RPi parsing. I will go through the code quickly now and look for possibilities to reduce memory use. I let you know.

jerabaul29, 2021-01-24:
Ok, a few tips for freeing a bit extra memory. Note that this will change some buffers sizes etc, so you should definitely test before deciding if things work. Are you well compiling from platformio, or from something else?

To get a bit extra available memory:

- Start from the code for the "working" Due code as usual: commit **39c30775849a** .

- The main place with potential for reducing the memory consumption is the serial buffers. As said, it is a bit tricky to reduce, and may have unexpected consequences, so you should double check carefully that things still work after. To reduce a bit the memory consumption, you should modify the ```platformio.ini``` file, at the root of ```LoggerWavesInIce_InSituWithIridium/DevelopmentCode/ArduinoMega/src```. There, the critical values are the ones in:

```
build_flags = -D SERIAL_RX_BUFFER_SIZE=512 -D SERIAL_TX_BUFFER_SIZE=512
```

You could try for example the following values:

- quite tough reduction:  ```build_flags = -D SERIAL_RX_BUFFER_SIZE=400 -D SERIAL_TX_BUFFER_SIZE=400```
- if the previous reduction does not work, can try a less hard reduction: ```build_flags = -D SERIAL_RX_BUFFER_SIZE=450 -D SERIAL_TX_BUFFER_SIZE=450```

Let me know how this goes. I think that the first solution has a good chance to work, and then you should get plenty of RAM back (on the order of slightly under 1KB extra available RAM). If it does not work fine, then the second one may do. If the second one does not work either let me know.

jerabaul29, 2021-01-24:
I just tested the configuration with size ```400```, it should free about 10% of the RAM. I think it *should* work fine, but let me know how tests go.

closed 2021-11-06 by jerabaul29